equinor/tagreader-python #7: Feature request - Tag metadata

einarsi, 2020-05-27:
Issue added on GitLab by @knudsvik 2018-06-12:

Include pull of metadata for collected tags. Especially tag description and units could be useful, but maybe range also?
if made available these data can be used for column headings, label descriptions etc.

einarsi, 2020-05-27:
Tag description and units for PI servers introduced in 5ae13364254ccbf443e75ec51aaa4921abaacba0 (get_units(), get_descriptions()) and extended to Aspen in d95fddac81985b2f77c7dedd3ed83c3f42ad78c5.

Can add range if I find the correct fields to fetch - feel free to point me in the right direction.

asmfstatoil, 2020-05-27:
https://github.com/equinor/PDP64/blob/master/PDP64/Query.cs

asmfstatoil, 2020-05-27:
du må bruke PI AF (evnt. PI rest api) for å få tak i korrekte verdiar for digital states

asmfstatoil, 2020-05-27:
når du kjem til Snorre må du til med MAPS, det var ikkje gøy, men har funne ein måte som fungerar ganske bra

einarsi, 2020-05-29:
Takk! Å hente ut IP.21 trenddata med mapping kan tagreader fra før, men her kan det se ut til at du har funnet en måte å også hente ut metadata for mappingene som ikke er default mapping? Veldig bra!

Det er fullt mulig å hente ut korrekte verdier fra digital states i PI med ODBC: be6029cc30f4402bb9f209a91f5475d2b8d89ceb

closed 2024-10-22 by mpuks