equinor/tagreader-python #34: Add option to return status

einarsi, 2020-09-17:
**Aspen IP.21 ODBC** returns a single status value which is 
  - 0: Good
  - 1: Suspect
  - 2: Bad
  - 4: Good/M (history has been modified)
  - 5: Suspect/M (history has been modified)
  - 6: Bad/M (history has been modified)

**PI Web API** can return threebooleans that may be relevant: 
  - Good
  - Questionable
  - Substituted
(Also Annotated, but probably not relevant here)

Assuming the other two backends can provide the same, an initial suggestion is to combine the three booleans from PI Web API into a numerical value corresponding to Aspen IP.21 ODBC nomenclature. Assuming that Good and Questionable are mutually exclusive: Status = Questionable + 2*(1-Good) + 4 * Substituted

torlid, 2020-11-07:
Went here to add this issue, but it was already posted :-)

Maybe a "collect Good data only" parameter could be one of several good solutions to solve this problem. 

Looking forward to having this new feature:

einarsi, 2021-01-26:
Aspen Web API: The field "Level" seems to map 1:1 to status from ODBC, at least for 0, 2, 4 and 6.

einarsi, 2021-01-27:
Seems as if PI ODBC has the following:
status (INT): 0 for good, can be both positive and negative
questionable (BOOL)
substituted (BOOL)

To conform with the above, I'll just replace any status value != 0 with 1.

einarsi, 2021-01-27:
Implemented functionality with PR #63

closed 2021-01-27 by einarsi