equinor/tagreader-python #35: Support for UNIX (REHL 7) [enhancement]

WestfalNamur, 2020-09-17:
Hi,

I would like to use tagreader on the UNIX. However tagreader (2.3.0) accesses the windows registry. This happens in  find_registry_key* and (I think) are used to . Are theses keys also stored on the UNIX machines so it would be possible to extend tagreader use to UNIX? 

All the best, Stefan

einarsi, 2020-09-17:
Hi Stefan,

I have never tried tagreader in a *nix environment before, but I would be happy to make it work it if it doesn't already.

The Windows registry is used to translate a data source (e.g. 'sna') to a server:port combination when basing queries on ODBC. Since ODBC is not an option outside of Windows due to proprietary drivers, you should use the Web APIs instead. The Web APIs should not need to touch the Windows registry at all (and if they do, then that's a bug).

Try to replace `pi` with `piwebapi` or `ip21` with `aspenone` when you call [tagreader.list_sources()](https://github.com/equinor/tagreader-python/blob/master/docs/manual.md#listing-available-data-sources) and [create clients](https://github.com/equinor/tagreader-python/blob/master/docs/manual.md#the-client). Does that work?

WestfalNamur, 2020-09-18:
Hi Einar,

first thank you or you support. After removing pi and pi21 from accepted_values in tagreader.list_sources() ("piwebapi", "aspenone" were already present) and removing  imports of winreg it worked. 
However I run into: an error: 
SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)

I am trying to solve this one but without success yet.

einarsi, 2020-09-18:
I would be very interested in details of which parts of tagreader code you needed to tear out to make it work. Would you be able to write a PR? Any changes made should not break Windows functionality. If not, then I can enter the wonderful world on *nix and work on it.

Regarding the SSL issue: You need to add an SSL key to your certifi certificate store. Have you tried the procedure described here? https://github.com/equinor/tagreader-python/blob/master/docs/manual.md#for-equinor-users
It works fine in Windows PCs, but untested for *nix environments.

WestfalNamur, 2020-09-22:
Hi, for now just the imports of winreg. Afterwards utils throws an error:
for cert in ssl.enum_certificates("CA"):
    AttributeError: module 'ssl' has no attribute 'enum_certificates'

closed 2024-10-22 by mpuks