equinor/tagreader-python #56: Use windows binaries for tables

vaasheim, 2021-01-15:
Collecting tables
  Downloading tables-3.6.1.tar.gz (4.6 MB)
     |¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦| 4.6 MB 6.4 MB/s
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\******\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\******\\AppData\\Local\\Temp\\pip-install-s96626t1\\tables\\setup.py'"'"'; __file__='"'"'C:\\Users\\******\\AppData\\Local\\Temp\\pip-install-s96626t1\\tables\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\******\AppData\Local\Temp\pip-pip-egg-info-4d2o5bx_'
         cwd: C:\Users\******\AppData\Local\Temp\pip-install-s96626t1\tables\
    Complete output (17 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\******\AppData\Local\Temp\pip-install-s96626t1\tables\setup.py", line 634, in <module>
        libdir = compiler.has_function(package.target_function,
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\distutils\ccompiler.py", line 792, in has_function
        objects = self.compile([fname], include_dirs=include_dirs)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\distutils\_msvccompiler.py", line 323, in compile
        self.initialize()
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\distutils\_msvccompiler.py", line 220, in initialize
        vc_env = _get_vc_env(plat_spec)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 314, in msvc14_get_vc_env
        return _msvc14_get_vc_env(plat_spec)
      File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\site-packages\setuptools\msvc.py", line 268, in _msvc14_get_vc_env
        raise distutils.errors.DistutilsPlatformError(
    distutils.errors.DistutilsPlatformError: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/
    * Using Python 3.9.1 (tags/v3.9.1:1e5d33e, Dec  7 2020, 17:08:21) [MSC v.1927 64 bit (AMD64)]
    * USE_PKGCONFIG: False
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

vaasheim, 2021-01-15:
Specific to python version 3.9 and newer

einarsi, 2021-01-15:
PyTables was last updated in October 2019, and apparently doesn't (yet?) provide wheels for Python 3.9 that can be used in Windows.

The default supported Equinor Python installation is 3.6, but tagreader is tested to work in 3.7 and 3.8 as well. I suggest you run tagreader in one of those Python versions instead.

closed 2021-01-28 by einarsi