equinor/tagreader-python #57: ImportError: cannot import name 'tables' from 'tagreader'

vaasheim, 2021-01-15:
```
import sys
sys.path.insert(0, "../")
import tagreader
import getpass
import tables

from tagreader.utils import add_statoil_root_certificate
from tagreader import list_sources

add_statoil_root_certificate()

#print(list_sources("aspen"))

c = tagreader.IMSClient("TRB", "aspenone")
c.connect()
#print(c.search(tag="tagname*", desc="**"))

tags = ['some-tag']
start_time = '01-JAN-2020 08:00:00'
end_time = '10-JAN-2020 08:00:00'
ts = 60

tagToPrint = c.read(tags, start_time, end_time, ts)

print(tagRead);

```

Traceback (most recent call last):
  File "C:\Users\******\Desktop\tagreader-python.py", line 5, in <module>
    import tables
  File "C:\Users\******\AppData\Roaming\Python\Python38\site-packages\tables\__init__.py", line 82, in <module>
    raise ImportError(
ImportError: Could not load any of ['hdf5.dll', 'hdf5dll.dll', 'pytables_hdf5.dll'], please ensure that it is installed in the package folder.

vaasheim, 2021-01-15:
Is this application only intended for servers without user security policies (admin users)? 
There should be a guid for which enviroments this application can run in without issues

einarsi, 2021-01-15:
Not quite sure I follow... tables (or rather: PyTables) is not a module or component of tagreader. However, it is a direct dependency, and if you have successfully and properly installed tagreader using 'pip install tagreader', then PyTables should be installed as well. Please elaborate on your environment and how you installed tagreader if that is not the case for you.

As to your second question: There are so many different environments that I can't possibly investigate, test and describe them all. Although I do admit that tagreader is not very instructive in its feedback when things fail, so finding the actual cause of an error can sometimes be a bit tricky. Can you be more specific as to what you are trying to do and how it fails for you? I would expect you to be able to use tagreader on any Equinor Windows PC that has access to the server you are trying to reach.

vaasheim, 2021-01-15:
Hi, I am sorry. Got a bit frustrated today when trying to get it working on a Equinor normal windows install for end users, I could not get tables working on 3.9 with this api might be an issue with tables and 3.9 or how this api use tables and updates in 3.9. Not sure
https://pypi.org/project/tables/
Could not get this working -> pip install tables do not work with 3.9 and 3.9.1, 3.8 was okay

Tagreader will only work on Equinor end users’ laptops (systems) with admin privileges. I have tested today and it was a pain to get working.
I several times had to use my administrator account do download and then copy dependencies over.
Since non admin users only have python 3.9 from Microsoft Store, they might have issues regarding the tables package

When I got it working it was a very good experience and easy to use 👍 

Just some code colleting tag data from AspenOne, maybe it could be useful in the documentation part somewhere, just add comments
`import sys
sys.path.insert(0, "../")
import tagreader
import getpass

from tabulate import tabulate
from tagreader.utils import add_statoil_root_certificate
from tagreader import list_sources

add_statoil_root_certificate()

#print(list_sources("aspen"))

c = tagreader.IMSClient("TRB", "aspenone")
c.connect()
#print(c.search(tag="TRB-PI*", desc="**"))

tags = ['TRB-FI -20-0615', 'TRB-FY -20-0612', 'TRB-FYQ -20-0612', 'TRB-FT -20-0616', 'TRB-LIC -20-0601']
start_time = '10-DEC-2020 12:00:00'
end_time = '10-JAN-2021 12:00:00'
ts = 60

tagToPrint = c.read(tags, start_time, end_time, ts)

print(tagToPrint);
f = open('TagsPrinted.txt', "a")
f.write(tabulate(tagToPrint, headers=['Time','TRB-FI -20-0615', 'TRB-FY -20-0612', 'TRB-FYQ -20-0612', 'TRB-FT -20-0616', 'TRB-LIC -20-0601']))
f.close()
`

einarsi, 2021-01-15:
I can definitely empathize with your frustration - I just spent the better half of yesterday reinstalling the AspenTech odbc drivers to allow me to perform today's update to 2.5.0.

It should be no problem to install tagreader without admin permissions. The only part that requires elevated privileges is installing odbc drivers, but those are strictly speaking not required anymore since you can use Web APIs for both PI and IP.21. 

Perhaps your primary python installation was installed using admin privileges? In that case, I can imagine adding packages may require elevated privileges as well. I can assure you that if you follow the recommended procedure for installing Python at Equinor, you will have no issues with normal user privileges: https://wiki.equinor.com/wiki/index.php/Software:Python#Python_on_Windows_in_Equinor

Glad to hear you enjoy using tagreader - I hope the initial wrinkles you experienced work themselves out.

BTW: I removed tags from your public post in accordance with IT security guidelines.

vaasheim, 2021-01-15:
I think you are correct regarding the issue I had with admin privileges, maybe it could be helpfull to include the wiki link in the documents aswell and a note that python version 3.9 (only up to 3.8) is supported (cause of wheels). I am at the back end support for IP21, we now only support ODBC driver installs if their needs is not covered by this tagreader API 🥇

closed 2021-01-28 by einarsi