equinor/tagreader-python #68: Overload str in client class to give a pretty print

asmfstatoil, 2021-03-15:
I want to see name and type of data server that client is connected to when calling print(client). Will make it fast and easy to pass connection information to loggers...

einarsi, 2021-04-28:
A pretty-printed string may not be the best option for something to pass along to other functions.
How about something like this instead?
client.status()
{"name": "Servername", "imstype": "aspenone", "imsbasetype": "ip21"}
(imsbasetype included since using Web API and ODBC should not matter)

You can format this however you want to your logger.

I suppose I could also define a __str__ method that uses the same information but with no requirements on format.

asmfstatoil, 2021-05-06:
str is the informal string representation of an object (and repr the formal one), it is a nice feature alongside the listing you specified. 

Primary use case for pretty print strings are debugging and logging for me.

closed 2024-10-22 by mpuks