equinor/tagreader-python #141: Reading non-numeric tags is NaN

LukeJakielaszek, 2022-11-16:
When reading tags that are not of numeric type, the tag is converted to a NaN value. This is due to line 489 of webhandlers.py and can be resolved by just removing the line. The line attempts to convert the tag to a numeric value, but it really should first check that the tag is numeric in the first place or allow the user to specify which tags are numeric.

```
# Ensure non-numericals like "1.#QNAN" are returned as NaN
df["Value"] = pd.to_numeric(df.Value, errors="coerce")
```

https://github.com/equinor/tagreader-python/blob/master/tagreader/web_handlers.py

franktoffel, 2023-03-03:
This seems good, has this been patched?

DWMorrow, 2023-08-11:
The only way I was able to overcome this is by using SQL extract.
**df = sql_query(tag, start_time, end_time)**

Using the experimental sql_query returns categorical values. Then use **c.read()** for numeric values. Not a solution, rather a workaround.

Asgmel03, 2024-10-22:
Hello,
I apologize for the (really) late reply from us. The Tagreader project has been shuffled around internally for a while, so no issues or PR's have been followed up for a while.

We have put this issue on our project backlog and will look into it soon. I will update you once we have a fix.