equinor/tagreader-python #202: Failing to install version 4.0.1 on mac m1 with python 3.11.2

asmfstatoil, 2023-06-08:
Downloading tables-3.8.0.tar.gz (8.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.0/8.0 MB 8.8 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [10 lines of output]
      ld: library not found for -lhdf5
      clang: error: linker command failed with exit code 1 (use -v to see invocation)
      cpuinfo failed, assuming no CPU features: 'flags'
      * Using Python 3.11.2 (main, Mar  8 2023, 10:10:03) [Clang 14.0.0 (clang-1400.0.29.202)]
      * Found cython 0.29.35
      * USE_PKGCONFIG: True
      .. ERROR:: Could not find a local HDF5 installation.
         You may need to explicitly state where your local HDF5 headers and
         library can be found by setting the ``HDF5_DIR`` environment
         variable or by using the ``--hdf5`` command-line option.
      [end of output]

closed 2023-06-09 by mortendaehli

mortendaehli, 2023-06-09:
@asmfstatoil, than you for the feedback. Are you running macOS?

This is an issue related to HDF5, used for caching, which is not installed "automagically" on macOS. I've removed the requirement when using ARM64 architecture, which is used by MacBooks the last 2+ years.

v4.0.3 has been released. Please report back if the problem persists.

Btw. if you just want to install HDF5 to use the cache functionality, you can follow the pytables guide here: https://www.pytables.org/usersguide/installation.html

asmfstatoil, 2023-06-09:
Tested and working.

PR #198 (2023-06-09): build(deps): don't install pytables for ARM64
I strongly suggest we don't do this. Rather we should explain the user that they need to install HDF5 in order for this to work on macOS/ARM64. I'm not sure, but I don't think HDF5 is default on any architecture and or OS, but you get HDF5 when you install scientific bundles/libraries such as the Anaconda bundle.

closes #202
closes #201