equinor/tagreader-python #253: Update certificate handling for Equinor

einarsi, 2023-09-01:
New certificates are being issued and the old ones will eventually expire. The existing solution for importing a specific certificate (the old root cert) specified by hash to certifi must therefore be updated. Perhaps amend `add_statoil_root_certificate()` to support both the old and the new certificate? 

I also suggest to consider adding functionality for fetching the certificate directly from pki.equinor.com. That would make things easier for Linux users and others using tagreader in environments without pre-existing Equinor certificates (e.g. Radix).

asmfstatoil, 2023-09-27:
I think the update of pi webapi certificates just broke tagreader and the add_statoil_root_certificate function

asmfstatoil, 2023-09-27:
#271

PR #271 (2023-09-28): feat: support new Equinor certificates
closes #253
closed 2023-09-28 by mortendaehli