equinor/tagreader-python #261: Time zones is not working

kjeel-dev, 2023-09-12:
Please see the below code. Time zones is not working, nor cache=None.
NB: Important to be able to specify timezone UTC!!!!!!

import tagreader
ftime = '2022-05-20 00:00'
ttime = '2022-05-20 04:00'

# This will read the data into df
c = tagreader.IMSClient('ims-pors-01', "piwebapi", cache=None)
df = c.read(['PI38090/Meas1/PRIM'], ftime, ttime, 60)

# This will crash trying to read the tag. Any string in tz will lead to a crash
c2 = tagreader.IMSClient('ims-pors-01', "piwebapi", cache=None, tz="Europe/Oslo")
df2 = c2.read(['PI38090/Meas1/PRIM'], ftime, ttime, 60)

# Moreover, cache=None does not work, it still creates the cache file

asmfstatoil, 2023-09-12:
can you try

[image]

as input tz when creating client?

Should be handled internally, but testing will make it easier to fix the issue

kjeel-dev, 2023-09-13:
Did a complete reinstall of my venv before testing (did not see your message). Don't know if that fixed it.

Now both a string and pytz.timezone works. I tested with tz="UTC", which worked (this is by far the most important, it should, in my opinion, be the default.

Regardless if I put cache=None when I create the client, or c.cache=None the cache directory is still created. This creates unwanted files all over, please fix ASAP.

The below warning is issued when creating the client. In my opinion, just turn of the caching as default at once. At least get it to work without caching and turn of the message.

[image]

Thanks,
BR.
Kjetil

mortendaehli, 2023-09-13:
I will create a new issue for the cache. This will be solved in the next release.

closed 2023-09-13 by mortendaehli