equinor/tagreader-python #298: Last valid point as read_type

kjeel-dev, 2023-12-07:
Demand for new functionality:

Example call:
df = c.read(['BA:CONC.1'], '05-Jan-2020 08:00:00', '05/01/20 11:30am', 60, read_type=tagreader.ReaderType.LastValid)

This should return data with a time step of 60 seconds. At each time step the data returned should be the last valid data point recorded before this.
Example: for the example call the first time step is 05-Jan-2020 08:00:00. The value given back should here be the last valid data point before this. This data point could be 1 second before or 1 month before, whatever is in the IMS.
The next time step is 05-Jan-2020 08:01:00. The value given back will be the same as for the first time step unless there is new valid time steps between 08:00:00 and 08:01:00. If there are several, the one closest to 05-Jan-2020 08:01:00 should be chosen.

hakmyo, 2024-09-10:
Agreed. I have also come across problems where this would be useful.

This function would be equivalent to pandas "forward fill" function: https://pandas.pydata.org/docs/reference/api/pandas.core.resample.Resampler.ffill.html#pandas.core.resample.Resampler.ffill

A workaround is to export RAW data from tagreader and then resample the data with pandas forward fill.

Asgmel03, 2024-10-22:
For the moment we do mostly maintenance and bug fixes for the Tagreader repo and do not implement new functionality at this time. But we have decided to put this issue into our board and discuss if we should look into it. I will update you once we decide what to do.