equinor/tagreader-python #302: Search with wildcard does not work on TRA

torlid, 2024-01-08:
Site = 'TRA', API = 'aspenone'
serach with no wildcard works:

ims.search('TRA-35PT3601A.PV')
[('TRA-35PT3601A.PV', 'Tog A gass innløp trykk')]

serach with wildcard does not:

ims.search("*-35TI*")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "c:\Users\torlid\OneDrive - Equinor\Python\pyCmpr\.venv\lib\site-packages\tagreader\clients.py", line 313, in search
    return self.handler.search(tag=tag, desc=desc, timeout=timeout)
  File "c:\Users\torlid\OneDrive - Equinor\Python\pyCmpr\.venv\lib\site-packages\tagreader\web_handlers.py", line 411, in search
    description = self._get_tag_description(tagname)
  File "c:\Users\torlid\OneDrive - Equinor\Python\pyCmpr\.venv\lib\site-packages\tagreader\web_handlers.py", line 461, in _get_tag_description
    data = self.fetch(url, params=query)
  File "c:\Users\torlid\OneDrive - Equinor\Python\pyCmpr\.venv\lib\site-packages\tagreader\web_handlers.py", line 146, in fetch
    return json.loads(txt)
  File "C:\Program Files\Python310\lib\json\__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "C:\Program Files\Python310\lib\json\decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Program Files\Python310\lib\json\decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

torlid, 2024-01-08:
tagreader          5.1.3

Asgmel03, 2024-10-22:
I guess this was solved by Åsmunds PR #340.
If this still is not working as expected, let me know and I will re-open the issue.

closed 2024-10-22 by Asgmel03