equinor/tagreader-python #326: The get requests to aspenone from BaseHandlerWeb.fetch does not work with Swedish characters

daha, 2024-03-12:
I can not fetch data for tags which contains Swedish characters. 

I get warnings like below and do not get out any data for requests like `c.read(tags=["åäö"],start_time=start_time, end_time=end_time, ts=1)`:

```
2024-03-12 16:07:59 WARNING  awt             API error for åäö: Tag Name %C3%A5%C3%A4%C3%B6 is invalid params: <Q f="d" allQuotes="1"><Tag><N><![CDATA[åäö]]><[/N](http://localhost:8889/N)><D><![CDATA[IP21]]><[/D](http://localhost:8889/D)><F><![CDATA[VAL]]><[/F](http://localhost:8889/F)><HF>0<[/HF](http://localhost:8889/HF)><St>1710169584000<[/St](http://localhost:8889/St)><Et>1710255984000<[/Et](http://localhost:8889/Et)><RT>1<[/RT](http://localhost:8889/RT)><S>0<[/S](http://localhost:8889/S)><P>10<[/P](http://localhost:8889/P)><PU>3<[/PU](http://localhost:8889/PU)><[/Tag](http://localhost:8889/Tag)><[/Q](http://localhost:8889/Q)>
```
 
 I have created my client like this:
 
 ``` python
c = tagreader.IMSClient(datasource="IP21",
                        url="http://my.domain/ProcessData/AtProcessDataREST.dll",
                        imstype="aspenone",
                        auth=auth,
                        tz=pytz.timezone("UTC"))
```

I have notices that when playing with the queries that is handles Swedish characters on post requests where the parameters as sent as data and not in the URL. So that kind of post requests seems to be more robust. I have verified that it works with post requests by hacking a bit in the tagreader code in  [web_handler.py  - BaseHandlerWeb.fetch](https://github.com/equinor/tagreader-python/blob/ec16d4610cfd2ab1f39c66ea14b96175b32360c5/tagreader/web_handlers.py#L139-L146) to something like below (but the sample code only works when params is a string). The encoding is needed to get correct size of the payload:

``` python
        res = self.session.post(
            url,
            data=params.encode("utf-8"),
            timeout=(
                None,
                timeout,
            ),
        )  # Noqa. Read timeout, No connect timeout.
```

daha, 2024-03-13:
I created some minor code changes which works better for me regarding Swedish characters in tags:
https://github.com/daha/tagreader-python/commit/8f7277a9c4adf17ffa4848be75c4e59565da1edf

Asgmel03, 2024-10-22:
Hello Daha,
I apologize for the late answer. Tagreader has been on the back-burner for a while as the responsibility for it has jumped around a bit internally.

Considering this software was initially built for Equinor who has all their tags in English, it might be an issue that went under the radar during development. It is not a priority fix for us, as we have to prioritize our tasks based on our budget, but I am happy you found a workaround. If we get some time to fix the issue in the future I will update you, until then I will close the issue.

closed 2024-10-22 by Asgmel03