equinor/tagreader-python #360: Tag optional or not?

smolvik1, 2024-06-19:
Perhaps a silly question, but is tag an Optional parameter in search as per the type hinting or not?
https://github.com/equinor/tagreader-python/blob/25743aca87f8cc0979569e756b2a445719185d16/tagreader/web_handlers.py#L444

smolvik1, 2024-06-19:
`c = tagreader.IMSClient("GFB", "aspenone")
c.connect()
c.search(desc='*Ringrom*')`

Yields Tag is required. 

Doing the same for a piwebapi client, e.g PINO, results are returned. Assuming then that the aspenone client requires the tag input when searching, whereas the piwebapi client does not.

Asgmel03, 2024-10-22:
I apologize for the late answer. Tagreader has been on the back-burner for a while as we have been tasked on different projects. Yes this is because Aspenone client requires the input tag when searching while the piwebapi does not. But I see that the typing is a bit confusing and I'll look into if we can find a better way to do it so that we don't get conflicting types.