equinor/tagreader-python #361: Installation not working with Linux

kjeel-dev, 2024-06-28:
I try to install tagreader using poetry add. I am on a Equinor RHEL8 TGX node. The error messages ends with:

Note: This error originates from the build backend, and is likely not a problem with poetry but with krb5 (0.5.1) not supporting PEP 517 builds. You can verify this by running 'pip wheel --no-cache-dir --use-pep517 "krb5 (==0.5.1)"'.

Running the following: pip wheel --no-cache-dir --use-pep517 "krb5 (==0.5.1)"' then leads to to errors taht ends with:

     gcc -pthread -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -Isrc/krb5 -I/private/kjeel/.cache/pypoetry/virtualenvs/python-quadbir-NGC_c9Ow-py3.11/include -I/usr/include/python3.11 -c src/krb5/_ccache.c -o build/temp.linux-x86_64-cpython-311/src/krb5/_ccache.o
      src/krb5/_ccache.c:39:10: fatal error: Python.h: No such file or directory
       #include "Python.h"
                ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for krb5
Failed to build krb5
ERROR: Failed to build one or more wheels

einarsi, 2024-06-28:
You probably need to install `libkrb5-dev` to be able to build the wheel. Not sure how that is done as regular user on TGX nodes. If you were on a Debian/Ubuntu system, it would be `sudo apt-get install libkrb5-dev`.

kjeel-dev, 2024-06-28:
Hmm, don't think that will be possible as a regular user. I can ask the admins, but the answer will probably be no.

closed 2024-06-28 by kjeel-dev